Makespace/WoodworkingWorkshop #5: Mitre saw needs new motor

mdales, 2019-04-30:


pinski1, 2019-04-30:
It is a 'Rage 3DB Mitre Saw' A new one is available [here](https://www.amazon.co.uk/Evolution-RAGE3-DB-Double-Multi-Purpose-Sliding/dp/B006BRVQ9K) for £246.00.

Wiki page [here](http://wiki.makespace.org/Equipment/MitreSawRage3Db).

Is it worth upgrading to a larger/more powerful/better saw? It's had a lot of use over the years and this may be a good excuse to upgrade.

camwoolf, 2019-05-07:
Has a new motor been order for the saw ?

mdales, 2019-05-10:
I don't believe anyone currently owns this task. I assume you've not done this @avdheever?

avdheever, 2019-05-10:
I have it on my list.  Place that does spares was meant to have a new motor
in this week.

mdales, 2019-05-14:
The replacement mitre saw is installed in place.

closed 2019-05-14 by mdales