Makespace/WoodworkingWorkshop #9: Setup test rig for Masso controller

mdales, 2019-05-03:
In the office we have a Masso CNC Router controller, the motors and motor drivers from the old test rig, and a new 24V PSU. 

We now need to put these together and get a working Masso demo going to judge its suitability to replace the DM500.

mdales, 2019-05-07:
I started this today, wiring up the Masso for the first time. However, I got blocked because the Masso ships without software - I need to send Masso the serial number and then I'll get access to the firmware.

@avdheever can you confirm that you've not already done this step, and if not that you're happy for me to do so (otherwise happy for you to do it, I can provide the serial number out of band).

avdheever, 2019-05-07:
I've not done this step - please proceed - they should send you SW for a 3
axis setup.  The warranty starts at that point .

Adrian

mdales, 2019-05-08:
Thanks. I've contacted their support with the serial number of the unit, I'll follow up once I have it and have updated the unit with the SW.

mdales, 2019-05-09:
I now have access to the software, so will try installing that today if I get chance.

mdales, 2019-05-10:
The Masso controller now has the correct software installed.

My next task is to hook it up to the test rig.

mdales, 2019-05-10:
I got the controller hooked up to the test rig in part - enough to start trying to get the device to home. I got the motor controllers working and the home sensors connecting, but it's still not homing properly. I'll do another check of the docs and then reach out to their support.

mdales, 2019-05-17:
I made some more progress on this today.

1. I got home working with a little fiddling, once I realised I had the directions set incorrectly for the z-axis. It's a little confusing still, in that although I've set a pull back distance and I observe it moving that pull back, the origin is set at the pull back position not (as other controllers have done AFAICT, at the position the switch is hit)
2. I wired up the z-probe, which is connected to +24V via a pull up resistor, and then earthed to ground when the bit touches. The most vexing thing here is that you have to enter g-code to do this [0]. Whilst this is certainly flexible (on the forums you see people writing nice double tap variants etc.), it means you need the probe g-code on the USB stick all the time AFAICT. This seems fragile in our environment.
3. The tool origin and work origin setup seem to be aimed more at the industrial workshop where these things are more constant. There's no (that I can see) one button X-Y origin and then Z probe as we're used to. You need to go to a page that manages a set of these configurations. Again, in our use case this seems a little fragile.
4. I tested a simple g-code generated from Fusion 360 using the supported Masso post processor [1], and that all worked well. Indeed, you get a nice preview of the design on screen before it is run, and where it is during running.
5. You have to set the X/Y axis feed rate max at below the machine max, as if you have the X/Y max feed rate at say 3000 mm/min and you say "go to work origin" then it will move slightly faster, and then the global setting of "max feed rate" kicks in and stops the motors. But the machine still thinks it is moving... Sigh. I guess the correct thing here is just to use the X/Y max rather than the global one.
6. The feed rate override I can't test, as it requires the pendant which @avdheever is setting up. However the forum discussions with their support team indicate it'll allow you to go between 20% and 100% of the feed rate.
7. I did test the soft limits, and if you load a g-code design that will go beyond the soft limits then it will not start the job (which is a failure of the DM-500 compared to our original controller).

Having got it this far I need to switch to some other tasks for now. Todo:

1. Properly understand the origin and tool setting, to see if we can make that workflow smooth enough for use in Makespace.
2. Test out g-code generated by vCarve Pro
3. Test out more complex designs that vary in z and another axis

[0] https://masso.com.au/masso-documentation/?section=touch-plate
[1] https://cam.autodesk.com/hsmposts?p=masso