Makespace/WoodworkingWorkshop #16: Investigate vCarve version on CNC Router computer

mdales, 2019-06-08:
Matt has asked on the google group:

> Is any chance to update the vcarve soft?  Existing version is from 2014 and is no way to upload or convert to 2014  sketchup files from 2019.

mdales, 2019-06-08:
I've asked Matt on the google group to clarify his use case, just so I understand what is required.

nickcadsoftuk, 2019-06-08:
FYI you can backsave from SketchUp 2019 right back to v3 via Save As... and then select version from dropdown.

sensor111, 2019-06-08:
> 
> 
> FYI you can backsave from SketchUp 2019 right back to v3 via Save As... and then select version from dropdown.

That's working. (I tried to use online sketchup which of course don't have that options)

sensor111, 2019-06-08:
> 
> 
> I've asked Matt on the google group to clarify his use case, just so I understand what is required.

Short description of the situation: I tried to do quick check of my idea on cncrouter and use chromebook with online sketchup. Even using walk around with 3D Warehouse it was impossible as it could be taken back to 3 latest version which is now 2017.
So, using the sketchup 2017 type of file vcarve just answer with short message: "it can't be open".
I assumed that reason is that vcarve is a version from 2014 and it can't deal with version 2017 sketchup.